juj/fbcp-ili9341 #330: raspberry pi zero w 1.3 lcd hat not displaying and giving error

pokedijimone, 2024-06-23:
i enter the command ./fbcp and get this error 
vc_dispmanx_display_open failed! Make sure to have hdmi_force_hotplug=1 setting in /boot/config.txt

[image]

and i have entered the setting hdmi_force_hotplug=1 and still gives the error

[image]

Edman274, 2024-07-03:
@pokedijimone, which version of Raspbian are you running?

green3554, 2024-07-08:
@Edman274 Having the same issue, using Raspbian 1:6.6.31-1+rpt1 (2024-05-29) armv7l / bookworm 32-bit with desktop. I've been screwing around with different settings and gotten nowhere except having the dstatistics show up I think.

pokedijimone, 2024-07-15:
> @pokedijimone, which version of Raspbian are you running?

the last version avalible on the raspberry imager raspberry pi zero w

anmolss111, 2024-08-21:
Use the legacy "Bullseye" os 32 bit.
run raspi-config to set
1. SPI Interface in Interface Options -> SPI -> Enabled
2. Enable the GL driver using Advanced options -> GL driver
3. reboot
4. build the fbcp with the -D driver on your LCD. I had a ST7789v with 320 and 240 aspect ratio so had to change C code refered in the issue.

bcw222, 2024-09-07:
comment the line `dtoverlay=vc4-kms-v3d` in /boot/config.txt, like this:
```
# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
```

bcw222, 2024-09-07:
Waveshare said it's not supported on bookworm.
Maybe try bullseye?

pokedijimone, 2024-10-13:
ok i changed to bullseye what shoukd i do now

anmolss111, 2024-10-13:
> Use the legacy "Bullseye" os 32 bit. run raspi-config to set

> 1. SPI Interface in Interface Options -> SPI -> Enabled
> 2. Enable the GL driver using Advanced options -> GL driver
> 3. reboot
> 4. build the fbcp with the -D driver on your LCD. 
I had a ST7789v with 320 and 240 aspect ratio so had to change C code refered in the issue.

pokedijimone, 2024-10-27:
how do i build fbcp

anmolss111, 2024-10-27:
You have do create a build folder in the directory you clone this repo to:

For eg.

cd ~
sudo apt-get install cmake
git clone https://github.com/juj/fbcp-ili9341.git
cd fbcp-ili9341
mkdir build
cd build

You can use many different arguments according to your requirements.

For eg.

# You need to put the driver of you LCD (in my case DST7789) and the DC and Reset GPIO pin number (software, not hardare pin numbers, generally mentioned in the usage manual of your LCD screen) 
cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=8 -DUSE_DMA_TRANSFERS=OFF -DSTATISTICS=0 ..

# Backlight control to switch off automatically after a while of inactivity
cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=8 -DUSE_DMA_TRANSFERS=OFF -DSTATISTICS=0 -DGPIO_TFT_BACKLIGHT=18 -DBACKLIGHT_CONTROL=ON ..

# Rotate the screen 
cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=8 -DUSE_DMA_TRANSFERS=OFF -DSTATISTICS=0 -DDISPLAY_ROTATE_180_DEGREES=ON ..

benoror, 2024-12-15:
In my case for [Waveshare 1.44inch LCD HAT](https://www.waveshare.com/wiki/1.44inch_LCD_HAT) I had to disable `#dtoverlay=vc4-kms-v3d` (https://github.com/juj/fbcp-ili9341/issues/311#issuecomment-1751712711) and compile with the following flags under Bullseye 32bit:

```
cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=20 -DWAVESHARE_ST7735S_HAT=ON -DUSE_DMA_TRANSFERS=OFF -DGPIO_TFT_BACKLIGHT=18 -DBACKLIGHT_CONTROL=ON -DSTATISTICS=0 ..
```

(via https://github.com/juj/fbcp-ili9341/blob/d0ebacf7c1f30b19b50997ebb67ba4f70ab95368/README.md?plain=1#L131)