juj/fbcp-ili9341 #333: rpi zero 2w + waveshare 1.3inch LCD HAT, can't display desktop

green3554, 2024-07-11:
I am able to get the display running and have the DSTATISTICS show up, however not the desktop. I believe that commenting the line "dtoverlay=vc4-kms-v3d" removes the desktop because I am unable to see it in realVNC. Though if I uncomment it, then I get an error that "vc_dispmanx_display_open failed! Make sure to have hdmi_force_hotplug=1 setting in /boot/config.txt" even though it is there. Any ideas? I've spent hours somewhat aimlessly trying things. 

OS: Raspbian 12 Bookworm, 32-bit with desktop. Kernel version is Linux 6.6.31+rpt-rpi-v7

Config file:
```
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
hdmi_group=2
hdmi_mode=87
hdmi_cvt=300 300 60 1 0 0 0
hdmi_force_hotplug=1
display_rotate=0
gpio=6,19,5,26,13,21,20,16=pu
```

bcw222, 2024-09-07:
See #330

> comment the line `dtoverlay=vc4-kms-v3d` in /boot/config.txt, like this:注释 /boot/config.txt 中的这一行 `dtoverlay=vc4-kms-v3d` ，如下所示：
> 
> ```
> # Enable DRM VC4 V3D driver
> #dtoverlay=vc4-kms-v3d
> ```

bcw222, 2024-09-07:
Please don't open a new issue with same problem.
See #330