juj/fbcp-ili9341 #335: Offset display

aaronalam1987, 2024-09-23:
Hopefully I am correct in my assumption that I can query this here, I am looking for a way to offset my display, everything is working great and I have managed to get it displaying in portrait mode (after fixing the flipped image)

However it is now center to the screen and I wish to push the main image to the top, I need to offset it negative 40 pixels I think. 

A point in the right direction would be appreciated.

[image]

Would like to move the image to the top.

aaronalam1987, 2024-09-24:
Nevermind, got it.

Modify line 67/68 of ili9341.h to suit.

67: #define DISPLAY_NATIVE_WIDTH 240
68: #define DISPLAY_NATIVE_HEIGHT 240

Still leaves me with a border across the top and bottom but nothing major.

PeterAJansen, 2025-04-13:
This was super helpful for me to figure out why a 320x320 ILI9488 was being offset -- the resolution was hard-coded to 320x480 and that was causing the black band at the top!

closed 2026-02-05 by aaronalam1987